eugr/Flashforge-for-Cura #2: Both Extruders heating for Single Extruder printing

ErroneousBosch, 2020-02-20:
I noticed that when I tell it to print single (right) extruder, it was heating up both extruders. Is this normal/expected, or should I re-setup my printer?

eugr, 2020-02-26:
@ErroneousBosch - no, it's not normal. Please make sure you disable unused extruder before slicing. If that doesn't help, the best way to get rid of the problem is to reset all Cura settings and set up the profile again.

One other user had this problem and resetting Cura settings helped. Possibly some leftover from previous experiments with different profiles.

ErroneousBosch, 2020-02-27:
Thanks for the insight. I ended up just disabling the extruder.

ErroneousBosch, 2020-02-27:
Also I can confirm that this works in 4.5 :D

closed 2020-02-27 by ErroneousBosch