eugr/Flashforge-for-Cura #4: Default printing temperature setting disabled

zackad19, 2020-04-12:
Hi there, 

Thanks so much for creating the profile. So far, it's worked great. I was trying to change the default printing temperature in Cura 4.5 but the setting wouldn't show up. It gave a message that said "This setting has been hidden by the active machine and will not be visible." Any chance you could help me figure out how to work around this? I usually like to print at a slightly higher temperature than the default, and I'd love to not have to change this setting each time I go to print.

Thanks for any help you can offer.

eugr, 2020-04-27:
This is weird... Are you changing the temperatures here (see picture below):
[image]

eugr, 2020-06-07:
Closing due to inactivity

closed 2020-06-07 by eugr