eugr/Flashforge-for-Cura #7: Heater in disabled nozzle heats up after initial layer.

J40yke, 2020-06-06:
Followed the instructions on GitHub. Installation went OK, and prints very well. Only problem is, even after disabling an extruder from the settings tab, that the disabled extruder will still warm up to the same temp as the active extruder. Even if the profile for the disabled extruder has a different temperature setting. Worth noting that the print starts correctly, with the disabled extruder cold. After the first layer has gone down that is the time that the disabled extruder heats up. I have to wait until the fault appears, then turn the heater off with octoprint.

J40yke, 2020-06-07:
Interesting thing happened tonight. Tried the exact same print, but with the left nozzle disabled, and the right nozzle enabled, and the disabled nozzle (left) stayed cold throughout the entire print...? Works as it should. I then enabled the left nozzle again, and disabled the right nozzle to run the print again. The right-side nozzle heated up after the left nozzle had printed the first layer again.

eugr, 2020-06-07:
This is weird.  Some people had that issue when they messed up with GCode scripts in the settings window. If you did that, I recommend resetting Cura to factory settings and reinstalling the profile again. If you edit any of GCode scripts through the interface that would override profile settings. GCode is generated dynamically in this profile.

J40yke, 2020-06-07:
Hi eugr. Thanks for getting back to me. This is my start gcode. All I changed was the vref max settings to suit new drivers I installed. High lighted below...

**** FlashForge Creator Pro PRINTER start.gcode for using left or right extruder ****
M83 ; relative extrusion mode
M73 P0 ; enable build progress
G162 X Y 1000 ; home XY maximum
G161 Z F1000 ; home Z minimum
G92 Z-5 ; set Z to -5
G1 Z0 ; move Z to 0
G161 Z F100 ; home Z slowly
M132 X Y Z A B ; recall home offsets
M135 T0 ; load right extruder offsets
{initial_pos_gcode, initial_extruder_nr}
G130 X20 Y20 Z20 A20 B20 ; lower stepper Vrefs while heating
M127; disable fan
M140 S{material_bed_temperature, initial_extruder_nr} T0 ; set bed temperature
M134 T0 ; stabilize bed temperature
M104 S{material_print_temperature_layer_0, initial_extruder_nr} T{initial_extruder_nr} ; set left or right extruder temperature
M133 T{initial_extruder_nr} ; stabilize left or right extruder temperature
**G130 X101.6 Y101.6 Z40 A101.6 B101.6 ; default stepper Vrefs**
G92 A0 B0 ; zero extruders
G1 Z0.4 ; position nozzle
M135 T{initial_extruder_nr} ; left or right tool
{nozzle_wipe_gcode, initial_extruder_nr}
G92 A0 B0 ; zero extruders
M73 P1 ;@body (notify GPX body has started)
; **** end of start.gcode ****

J40yke, 2020-06-08:
So I followed your advice. I completely removed cura 4.6, and reinstalled with a fresh set of definitions, meshes and extruder files. My new stock start gcode is below. Exactly the same problem as before.

; **** FlashForge Creator Pro PRINTER start.gcode for using left or right extruder ****
M83 ; relative extrusion mode
M73 P0 ; enable build progress
G162 X Y F3000 ; home XY maximum
G161 Z F1200 ; home Z minimum
G92 Z-5 ; set Z to -5
G1 Z0 ; move Z to 0
G161 Z F100 ; home Z slowly
M132 X Y Z A B ; recall home offsets
M135 T0 ; load right extruder offsets
{initial_pos_gcode, initial_extruder_nr}
G130 X20 Y20 Z20 A20 B20 ; lower stepper Vrefs while heating
M127; disable fan
M140 S{material_bed_temperature, initial_extruder_nr} T0 ; set bed temperature
M134 T0 ; stabilize bed temperature
M104 S{material_print_temperature_layer_0, initial_extruder_nr} T{initial_extruder_nr} ; set left or right extruder temperature
M133 T{initial_extruder_nr} ; stabilize left or right extruder temperature
G130 X127 Y127 Z40 A127 B127 ; default stepper Vrefs
G92 A0 B0 ; zero extruders
G1 Z0.4 ; position nozzle
M135 T{initial_extruder_nr} ; left or right tool
{nozzle_wipe_gcode, initial_extruder_nr}
G92 A0 B0 ; zero extruders
M73 P1 ;@body (notify GPX body has started)
; **** end of start.gcode ****

eugr, 2020-06-08:
Looks good. It’s a single extruder version, it will be different when both extruders are active. Could you save your project as 3mf file and attach here? Or attach generated .gcode file (not x3g)?

I’ve also realized that I’m using newer version of the profile that I was preparing for submission to Cura repository, probably I need to push the changes here too. Not much changed, but I streamlined GCode generation a bit.

J40yke, 2020-06-08:
I'm not able to attach files other than JPEG, PDF etc etc.

J40yke, 2020-06-08:
[ffcp test.zip](https://github.com/eugr/Flashforge-for-Cura/files/4745223/ffcp.test.zip)

J40yke, 2020-06-08:
I'm hoping I've uploaded what you asked for. I've hopefully also included my current printer profile.

eugr, 2020-06-14:
Sorry for late reply, too busy at work!

Weird, when I open your 3mf file it is set to print on right extruder only.
BTW, how do you print using my profile? Do you use Octoprint or save to SD card?

Let's try something different. Please slice your file using my profile and save it as .gcode file (not .x3g!) and attach it here. I can then look at generated code and maybe I could spot an issue then.

J40yke, 2020-06-15:
That's odd? Because that file definitely prints on left nozzle, and warms the right nozzle after the initial layer. I tried it before i uploaded it to make sure.
From Cura, I print via the octoprint plugin. I click the "print with octoprint" button. I've attached the sliced file from Cura (latest version).  I appreciate your help with this.
[FCPRO_xyzCalibration_cube.zip](https://github.com/eugr/Flashforge-for-Cura/files/4782937/FCPRO_xyzCalibration_cube.zip)

J40yke, 2020-07-10:
Hi. Any updates on this?

CactusHamster, 2020-12-15:
Just a quick idea I had: In Cura's expert slicing settings there is an option labelled "Standby Temperature" under "Material" that is set to 150 at default. Maybe setting this to 0 will keep the unused extruder from heating up?

J40yke, 2020-12-15:
Hi, thanks for the reply. Where abouts is that setting? I can’t find it?

James

J40yke, 2020-12-15:
Found it!! Will try this and report back. It’s also worth mentioning the stepper motor gets very hot on the unused extruder if only using one nozzle to print. To overcome this I disconnect it.

James

eugr, 2020-12-15:
@J40yke - hey, sorry, I looked at your GCode and don't see anything there that would warm up the right nozzle. Haven't tried printing it on my printer, but I'm pretty positive it would work just fine.

I'd look into your Octoprint settings. You need to make sure you don't have ANY start/stop GCode in the printer settings, except for After print job is cancelled (that's optional). You need to remove any other GCode from there (be sure to scroll down):

[image]

J40yke, 2020-12-15:
Just double checked my octoprint settings. There’s nothing in any of the gcode boxes. They’re all blank.

James

[image]<https://user-images.githubusercontent.com/2192184/102281997-c5fba100-3ee4-11eb-9b47-8db3fdf65c19.png>

eugr, 2020-12-15:
This is weird. Can you check the printer profile and make sure "shared nozzle" in "hotend & extruder" is not checked?
[image]

Also that GPX is set correctly:
[image]

I'd also try (if you haven't yet) to upload generated GCode file to Octoprint manually (vs. send to octoprint) and initiate printing from Octoprint interface.

J40yke, 2020-12-16:
OK, so apart from gcode flavour, my setup is the same as yours. Mine was set to reprap. I’ve now changed it to MakerBot. Will have a go and let you know. Thanks again.

James

[image]<https://user-images.githubusercontent.com/2192184/102285186-964f9780-3eea-11eb-9052-7cf492e5e83a.png>

[image]<https://user-images.githubusercontent.com/2192184/102285315-eb8ba900-3eea-11eb-819b-1fdb5a19e703.png>

slm020, 2021-02-15:
Hi, I am having this problem - I'm very new to this, but I have found that if I remove the "M104 S200 located about three lines before the mesh section, the right doesn't heat up.

```
G1 F900 X-9.574 Y-11.837 E0.08191
G0 F1800 X-9.685 Y-11.665
**M104 S200**
G1 F900 X-9.528 Y-11.508 E0.00554
G1 F1800 E-1.3
;MESH:NONMESH
G0 F600 X-9.528 Y-11.508 Z0.45
```

J40yke, 2021-02-15:
> Hi, I am having this problem - I'm very new to this, but I have found that if I remove the "M104 S200 located about three lines before the mesh section, the right doesn't heat up.
>
> ```
> G1 F900 [...]
This is my FFCP start up gcode in CURA. looks like I don't have the same as you? mine is standard.
; **** FlashForge Creator Pro PRINTER start.gcode for using left or right extruder ****
M83 ; relative extrusion mode
M73 P0 ; enable build progress
G162 X Y F3000 ; home XY maximum
G161 Z F1200 ; home Z minimum
G92 Z-5 ; set Z to -5
G1 Z0 ; move Z to 0
G161 Z F100 ; home Z slowly
M132 X Y Z A B ; recall home offsets
M135 T0 ; load right extruder offsets
{initial_pos_gcode, initial_extruder_nr}
G130 X20 Y20 Z20 A20 B20 ; lower stepper Vrefs while heating
M127; disable fan
M140 S{material_bed_temperature, initial_extruder_nr} T0 ; set bed temperature
M134 T0 ; stabilize bed temperature
M104 S{material_print_temperature_layer_0, initial_extruder_nr} T{initial_extruder_nr} ; set left or right extruder temperature
M133 T{initial_extruder_nr} ; stabilize left or right extruder temperature
G130 X127 Y127 Z40 A127 B127 ; default stepper Vrefs
G92 A0 B0 ; zero extruders
G1 Z0.4 ; position nozzle
M135 T{initial_extruder_nr} ; left or right tool
{nozzle_wipe_gcode, initial_extruder_nr}
G92 A0 B0 ; zero extruders
M73 P1 ;@body (notify GPX body has started)
; **** end of start.gcode ****

closed 2021-02-15 by J40yke

slm020, 2021-02-15:
Ah, no, the code M104 is not in the printer section, it's way down, looks like something that Cura generates. If you do a search on the attached gcode for M104 - you will find one in in the start section but also one just in the Skins portion before Meshes. I think the original OP was having the same problem, and I am not able to figure out how to remove it without editing the gcode each time unfortunately.
[Lid.txt](https://github.com/eugr/Flashforge-for-Cura/files/5984410/Lid.txt)