eugr/Flashforge-for-Cura #8: Why does the GCODE wait for the bed to heat then follow with the extruder.

flubbacrud, 2020-06-30:
Start GCODE waits for bed then starts heating extruder. Why not do both simulateously? Line 14-17

`M140 S{material_bed_temperature, initial_extruder_nr} T0 ; set bed temperature
M134 T0 ; stabilize bed temperature
M104 S{material_print_temperature_layer_0, initial_extruder_nr} T{initial_extruder_nr} ; set left or right extruder temperature
M133 T{initial_extruder_nr} ; stabilize left or right extruder temperature`

A simple change to:

`M140 S{material_bed_temperature, initial_extruder_nr} T0 ; set bed temperature
M104 S{material_print_temperature_layer_0, initial_extruder_nr} T{initial_extruder_nr} ; set left or right extruder temperature
M134 T0 ; stabilize bed temperature
M133 T{initial_extruder_nr} ; stabilize left or right extruder temperature`

maathieu, 2020-09-19:
Hey, I'm using a Flashforge Creator 1 and for my use case this is good: the print bed heats up much more slowly than the extruder(s), so with a tool that fires both at the same time, I end up getting a lot of oozing before the print begins, while the gcode waits for the print bed to reach temperature. I think this is a feature ;).

eugr, 2020-12-04:
Yes, this is a feature to avoid excessive oozing, not a bug. Closing this issue.

closed 2020-12-04 by eugr