eugr/Flashforge-for-Cura #9: First layer has an Y offset when printing with dual nozzle

abelsilva, 2020-07-15:
When I print with dual nozzle, the first layer of the first printing nozzle has an Y offset

Is there a setting that could cause this?
If not, Is there a way I can provide any information to help diagnose?
Or is this not at all related to the profile?

Picture shows a raft printed in PVA to showing the first layer with an offset
[image]

mkepo, 2020-09-01:
Hi, could you solve the problem? I am facing this issue too, and couldn't find the solution.

abelsilva, 2020-09-01:
No, tried changing almost everything in settings and start script, including

* added Y travel during wipe
* added 2x Z travels, one with each extruder active
* tried not changing vrefs 

The shift only happens when changing to the 2nd extruder on the first layer
Then everything keeps printing using that shift (or maybe it's only the first layer from the first extruder that has the shift)

I don't get this using Flashprint, but would prefer to use Cura as it has many settings not available in Flashprint

jrwwallis, 2020-09-13:
I've been experiencing y-axis skips when using the Cura/eugr/FFCP combo too.  For me it can happen with just one extruder.  Using FlashPrint, I don't get any y-skips.

The fix I found was to comment out the following two lines in the startup gcode:

G130 X20 Y20 Z20 A20 B20 ; lower stepper Vrefs while heating

G130 X127 Y127 Z40 A127 B127 ; default stepper Vrefs

Go to Preferences -> Printers -> Flashforge Creator Pro -> Machine Settings
In the Start G-code, comment out the above two lines by putting ; semicolons at the beginning of each line.

The 1st line turns down the stepper motor voltages during extruder heating, then the 2nd line is supposed to put the voltages back to default values (127).  However, it appears the FFCP interprets 127 as max voltage instead of default voltage.  Turning up the voltage too high causes the steppers to overheat and skip.  Commenting out these two lines keeps the voltages at default levels the whole time, which fixes the y skipping issue for me.

In addition to the above, it may be necessary to powercycle or even factory reset your FFCP - I don't know if the G130 voltage settings are persisted or not between prints or powercycles.

abelsilva, 2020-09-13:
Thanks, doesn't fix, tried that already, just confirmed now with factory reset also, same issue

I don't get "y-axis skips"
I get a single y-axis skip, always on the same moment, after printing first layer of the first extruder

Unex87, 2021-03-05:
Hello, any fix found yet. mine is doing same thing!

loshlee, 2021-04-13:
Are you all using Macs? If so, I think this is the kind of thing I would want to try a clean system for. There are a lot of things about text encodings, sysprefs, extensions, permissions that can still go slightly askew. With the upcoming systems that Apple is expected to introduce, we may lose the ability to boot externally, but for now, that's what I'd try. And everybody can use a backup system drive, anyway. Create a clean OS on an external drive or clean partition of your main drive doing as little customization as possible, boot from it, then install Cura and FlashForge_for_Cura. If this problem follows that installation, then it's likely something about the config that hasn't yet been revealed. But there's got to be some reason I have a Mac installation of this thing that works perfectly, while it doesn't work for others.

jmj3886, 2022-01-21:
This may help you, I was also having a similar or the same issue. When T1 began extruding, after N-layers using T0, the entire print would shift ~-2mm on the Y axis. I looked everywhere for a solution, moving my model, moving my prime tower, removing comments from the gcode incase GPX couldn't handle them. 

I finally found a fork of this repo, https://github.com/sesca128/Flashforge-for-Cura/commit/58f69012d05b6cdcf5f4615fa62cb2c4c0f032fd, that restricted the Y-axis to 145mm. This solved my issue. I am not entirely sure why, so if someone is able to elaborate, that would be wonderful. But setting the Y-axis to 145mm fixed my ~-2mm Y-axis shift on the first tool change.

loshlee, 2022-01-21:
Thanks. First time I tried dual extrusion with it, the tool change printed maybe 10 mm off on the y axis. I don't think I'm going to be able to solve that. There also seem to be some minute compromises to print quality for high resolution prints that I don't encounter using FlashPrint. Same is true for the  DrLex0 configurations for PrusaSlicer. I think I've just about decided to give up on the other slicers as long as I'm using this printer.