eugr/Flashforge-for-Cura #15: Cura 4.8 update difficulty

loshlee, 2020-11-19:
After upgrading to Cura 4.8, some default preset settings appear to be retained even after editing for a one-off print. For example, temperature settings for my PLA preset were 200°C and I created an X3G file after changing to 210°, the printing started and stayed at 200. Haven't yet observed whether this happens if I create a new preset. I suppose this could also be the X3GWriter extension's fault.

loshlee, 2020-11-26:
I reverted to Cura 4.7.1 and the problem still existed, so deleted the Cura folder in Application Support and started from scratch. The problem no longer exists. I'll report back if installing 4.8 brings the problem back.

loshlee, 2020-11-26:
Because the problems remained after reverting to Cura 4.7.1 the first time, after a clean installation made the problem disappear for 4.7.1, that the problems would not recur in 4.8. However, the problems came back when I tried 4.8 again. This time reverting to 4.7.1 fixed the problems without a clean install.

eugr, 2020-12-04:
For some reason I'm not getting any notifications from GitHub... Anyway, I've just installed 4.8 - will have a look.

loshlee, 2020-12-05:
Thanks. I've begun to believe that the problem may be related to the working order of assigning the nozzle if the print is printed entirely on one nozzle. It may be that one enabled nozzle will let the print commence in 4.8 even if the item printed is not pre-assigned to that nozzle, thereby bypassing some settings. I haven't confirmed this, but one of the prints I made did work correctly regarding custom filament settings. Thanks again.

eugr, 2020-12-07:
In general, when printing with a single nozzle, it's always best to disable another one to avoid any issues. It's been a while since I worked on my profile, but I remember that there was no reliable way to tell if any given extruder is actually used anywhere in the print, so if both extruders are enabled, it would assume that you are doing double extrusion and use the corresponding GCode. Also, IIRC skirt/brim is by default assigned to the right extruder in Cura.

So, yeah, if not using dual extruders, always disable the unused one.

Going to close this ticket now.

closed 2020-12-07 by eugr

loshlee, 2020-12-08:
Thanks. The only issue with that assumption is that I always disable one extruder if the print is done on a single extruder, and always have. Maybe I didn't disable the unused extruder until after the model was loaded. I've not confirmed it, but if that's the case, then the issue I experienced may be related to working order.

eugr, 2020-12-08:
As long as you disable the unused extruder before slicing, it should work properly. I don't think it makes any difference when loading the model. Could be some glitch with 4.8 though.