eugr/Flashforge-for-Cura #18: Prints begin by printing material off of the print bed

AryavP, 2021-03-11:
When I start a print, the printer first extrudes material from the right nozzle off of the right side of the print bed, and then the left nozzle off of the left side. All the objects I'm trying to print are on the print bed in Cura when I export the print file.

loshlee, 2021-04-12:
I think this is known as the Prime Blob, and I'm trying to figure out if there's a way to make the setting accessible for this config.

[image]

There seems to be a [related issue](https://community.ultimaker.com/topic/29917-help-setting-not-visible/) discussed in the Ultimaker forums.

loshlee, 2021-04-12:
According to Lyl3, whose [project](https://www.thingiverse.com/thing:3569113) at Thingiverse is one of the sources for eugr's config, the nozzle preparation step is a result of constructs for "reposition nozzle", "purge nozzle", "slow wipe", and "lift". These may be related to lines 17 thru 23 of our Start G-Code, but I'm not very close to figuring out exactly how to change it for a nozzle purge that I can live with.

loshlee, 2021-04-12:
Well it appears that I was able to successfully eliminate the "blob" or "cleaning strip" or "nozzle wipe" simply by commenting out line 22 of the Start G-Code (Preferences>Printers>[select printer]>Machine Settings). That is, by changing:

{nozzle_wipe_gcode, initial_extruder_nr}

to

****{nozzle_wipe_gcode, initial_extruder_nr}

Duh, hope nobody else has published this discovery and I just failed to find it. Wouldn't surprise me too much, though, I ignore the obvious way too much to be a professional programmer. Still, I guess it's not a good idea to assume you understand what the code precisely means before doing a little research. Without the "nozzle wipe", you should be able to manage your nozzle preparation by adding a Skirt (Build Plate Adhesion in Print settings), the way FlashPrint does.

I suppose I could try to find the parameters for "nozzle_wipe_gcode" in the source files, but I think this should suffice for now.

loshlee, 2021-04-29:
I noticed that, although the "cleaning strip" wasn't printed after making the above change, a "blob" was instead printed in the front corner, but only for prints assigned to the left nozzle. I guess I'll need to compare nozzle code to find out how to eliminate that.

Meins321, 2021-07-15:
then it might be one of the commands located under extruder which makes the nozzle purge....

But why do you remove it? it wasted just a small amoung of filament and is a very usefull feature for not having any blob on the print if you dont print with brim or raft...

+loshlee
Do you happen to know how to drive the extruder head slower to the starting point in the front right corner?
With this current profile  the printer drives very violent to the bed after heatup of the nozzel&bed and then does the snot purge..

Maybe that is something we should atleast halve as sparing 1second is not worth the meachnic wear of the head

loshlee, 2021-07-15:
> +loshlee
> Do you happen to know how to drive the extruder head slower to the starting point in the front right corner?
> With this current profile the printer drives very violent to the bed after heatup of the nozzel&bed and then does the snot purge..
> 

https://www.thingiverse.com/thing:3569113/comments#comment-5292436