eugr/Flashforge-for-Cura #19: Cura 4.9 introduces support for FlashForge Dreamer NX

loshlee, 2021-04-21:
Is this the first time a FlashForge printer has been supported by Cura. Anyway, the Application Support files for Cura 4.8 seem to have been migrated to the 4.9 folder, because the Creator Pro still exists in the list of printers, and I can still print successfully. However, I can't seem to find the Creator Pro in the list of printers to add again, even after reinstalling Flash-forge-for-Cura. I wondered if support for the Dreamer NX may have interfered with the current installation process.

bhast2, 2021-04-21:
I just rename the dreamer nx definitions file to .old and the creator pro showed up

loshlee, 2021-04-21:
OK, thanks. I'm guessing eugr can make a change to the creator pro definitions file so that the two will be compatible, but I'm just guessing. The problem could lie somewhere else entirely, AFAIK.

m0illi, 2021-04-22:
I also experienced this issue. Adding the files to a fresh install of 4.9, Creator Pro didn't show in the list of printers.

loshlee, 2021-04-25:
I thought maybe I had it when I read [this](https://www.thingiverse.com/thing:3569113/comments#comment-5405479), but I made the change to creator_pro.def.json and relaunched Ultimaker Cura, but it didn't seem to make any difference.

m0illi, 2021-04-25:
That might have something to do with it - because I was able to get it to work by deleting the pre-packaged "Dreamer" files out of my install before adding in the creator pro files.

yoshadow, 2021-04-27:
> I just rename the dreamer nx definitions file to .old and the creator pro showed up

I didn't find that file in **linux** so this is what I did.

1. Go to /home/user/.local/share/cura/4.9/definitions
2. Open the **creator_pro.def.json** and change the _manufacterer_. (I changed to CreatorFF)
3. Save the file.
4. Reload Cura.

That's it, you can find the printer like CreatorFF.

loshlee, 2021-04-27:
I did a little more hacking. 

1. I quit Ultimaker Cura 4.9. 
2. I took the **creator_pro.def.json** out of the definitions folder. 
3. I made a copy of **flashforge_dreamer_nx.def.json** in the definitions folder and renamed it to **flashforge_creator_pro.def.json**.
4. In **flashforge_creator_pro.def.json** I changed "FlashForge" to "Flashforge" (making the second "f" lower case).
5. I converted the **FlashForge_CreatorPro.stl** in the **meshes** folder to **FlashForge_CreatorPro.obj** and replaced **FlashForge_CreatorPro.stl** with the **FlashForge_CreatorPro.obj** in the **meshes** folder.
6. In **flashforge_creator_pro.def.json** I changed "**FlashForge_CreatorPro.stl**" to "**FlashForge_CreatorPro.obj**".
7. OMM, now when beginning the process of adding a printer in settings Ultimaker Cura 4.9, I can see both "**Dreamer NX**" and "**FlashForge Creator Pro**" under the **Flashforge** heading.

Haven't tried to actually add it yet, because my existing instance for FFCP is still working OK. For now I'll let someone else test if they care to. Please let me know if you find that it works. Thanks.

loshlee, 2021-04-27:
I left step 4 out of the first reply (sorry), so if you're reading an email or other notification, please visit the site (https://github.com/eugr/Flashforge-for-Cura/issues/19#issuecomment-827983060).

loshlee, 2021-04-28:
I eventually changed line 3 of **flashforge_creator_pro.def.json**  to

"name": "Creator Pro",

to reduce redundancy, and I eventually added the printer without issue. In MacOS Mojave the listing appears as:

[image]

Valkyrie1999, 2021-06-02:
> I just rename the dreamer nx definitions file to .old and the creator pro showed up

This fixed my issue

loshlee, 2021-07-05:
> > I just rename the dreamer nx definitions file to .old and the creator pro showed up
 But wouldn't you rather have the option to choose either Creator Pro or Dreamer NX? Anyway, from what I can tell so far, the hack I mentioned above holds up through the 4.10 upgrade.

Meins321, 2021-07-14:
Yes right on a fresh install of Cura 4.10 you need to go into

C:\Program Files\Ultimaker Cura 4.10.0\resources\definitions\ flashforge_dreamer_nx.def.json **.old**

Maybe change something in the provided files that the creator pro shows up alongside the nx?

Its now not possible to install currently without findign this!

philippe44, 2021-07-30:
IMHO the real culprit is that the "manufacturer" key is "Flashforge" in the built-in JSON manifests (in \Program Files...) and "FlashForge" in this extension. Just rename them consistently using either 'f' or 'F' and they both appear. I guess the JSON parser is confused by the two FlasfF(f)orge" and some part is case-sensitive, some is not

KillerJOhn64, 2021-09-28:
Hey guys, 
I've tested what @loshlee said with json files and now my Cura 4.11, recognizes the Creator Pro, but when I click Add the software crashes.

KillerJOhn64, 2021-09-28:
I also, just tried it now with Cura 4.8 and it happens the same - crashes when I hit next

loshlee, 2021-09-28:
I just successfully added a second instance after upgrading to 4.11. Do you already have another printer installed? You might try adding a different printer first. I may have to try it for a user or even a system where it hasn't been installed before to see if I can replicate the issue. You might try creating a different user. I've found Cura to be quite adaptable to customization, but there's no assurance that it always will be as newer versions are introduced.

loshlee, 2021-09-28:
> I also, just tried it now with Cura 4.8 and it happens the same - crashes when I hit next

My suggestions weren't intended to work with 4.8, eugr's method suffices for that version.

goldencanine, 2021-11-09:
Loshlee I'm getting the same issue with crashing on cura 4.11 , deleted and reinstalled and double checked definitions but it keeps crashing :(

loshlee, 2021-11-10:
I'm having problems with creating a printer as well. It's not crashing for me, but a new instance of the FFCP fails to draw the base. Most of the base models (meshes folder) are .obj, but some are .stl. I haven't discovered what parameters are involved in fixing that issue yet.

goldencanine, 2021-11-10:
@loshlee I believe I have it figured out, running Cura 4.12 . I'll run a test print and upload my files to you later
I kept having issues with Eugrs file so I assume they changed something about definition syntax, and instead copied over key qualities to the included dreamer nx profile, renamed it, and linked to the proper mesh.

eugr, 2022-01-16:
Sorry guys, haven't paid attention to this repo for a while. I've merged a pull request fixing this issue

closed 2022-01-16 by eugr