eugr/Flashforge-for-Cura #21: Gcode for Octoprint

PeterJBurke, 2021-05-10:
Request:
Enable gcode generation that works with Octoprint.
Using gcode generated from the "Dreamer NX" printer definition in CURA 4.9, I can load the gcode to Octoprint and it prints on the Flashforge dreamer.

Using gcode generated from the "Creater Pro" printer definition (this repo) in CURA 4.9, I can load the gcode to Octoprint and it gives  a "file error" on the Flashforge Dreamer printer.

My system:
uname -a
Linux i9GarageV2 5.4.0-72-generic #80-Ubuntu SMP Mon Apr 12 17:35:00 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Cura 4.9
XGwriter installed

Octoprint
Version 1.4.0

Flashforge Dreamer

loshlee, 2021-05-13:
Have you tried GPX (gcode to x3g converter, elsewhere here at GitHub)? I wonder if you might be able to customize the conversion routine for the gcode to eliminate the file error.

Hammerhead753, 2023-04-24:
> Request: Enable gcode generation that works with Octoprint. Using gcode generated from the "Dreamer NX" printer definition in CURA 4.9, I can load the gcode to Octoprint and it prints on the [...]

You need to uninstall the x3gwriter from Cura, then add the GPX Plug in to Octoprint https://plugins.octoprint.org/plugins/gpx/
by doing this you are having Octoprint convert the gcode to x3g instead of your  slicer.  This works, this is how I do it.  You won't see anything in Octoprint to confirm it has been converted, just upload the gcode file and print.  This method also allows you to use a different slicer, I'm still experimenting with using this machine on Prusaslicer.