eugr/Flashforge-for-Cura #24: No Luck With Flashforge Creator Max

OGYardens, 2021-09-30:
Hi there!  I dropped the files in both file locations with admin access on my laptop and Cura does not provide an option for the Creator Pro. Just Dreamer NX.  

File drop locations:  
C:\Users\[user name]\AppData\Roaming\cura\4.11
C:\Program Files\Ultimaker Cura 4.11.0\resources

I verified the relevant files were successfully copied to the file locations.  Tried one, then the other.

UdoMeessen, 2021-12-03:
Hi, the problem is solved. Look at that issue:

https://github.com/eugr/Flashforge-for-Cura/issues/25 "Creator Pro does not show up"

PHiGGYsMALLS, 2021-12-04:
> Hi, the problem is solved. Look at that issue:
> 
> #25 "Creator Pro does not show up"

Thank you for notifying me, but it didn't solve it for me.

PHiGGYsMALLS, 2021-12-04:
> Hi, the problem is solved. Look at that issue:
> 
> #25 "Creator Pro does not show up"

Only shows up after adding in the AppData\Roaming profile.  Thanks for the update, @UdoMeessen

UdoMeessen, 2021-12-04:
Yes, it's strange...
I have installed several generations of Cura on the same computer under Win10.
Once the package has to be installed in AppData/Roaming and another time directly after Program Files. There seem to be different internal algorithms depending on the distribution.

But MAIN THING, it now works for you too, @PHiGGYsMALLS