eugr/Flashforge-for-Cura #25: Creator Pro Does not show up

noamtsvi, 2021-11-22:
I copied the files into the folders, but Creator Pro still doesn't show up in the configurations.

gustavmangion, 2021-11-24:
I've faced this issue myself, but its an easy fix.
All you need to do is to edit creator_pro.def.json in the definitions folder. I would recommend using Notepad++ and not Notepad. Just change the name of the manufacturer, I set mine to "Flashforge Custom" instead of "Flashforge" and I can see it in Cura. 
Hope this helps

noamtsvi, 2021-11-25:
I've already found another setup by Toylerrr, works great. but thanks.

noamtsvi, 2021-11-28:
I tried this and now it works great, thanks!

mlw113, 2022-01-08:
I tried the edit suggested by gustavmangion from nov 2021.  That did not work for cura 4.12  Appreciate help.  TY.

duckinator, 2022-05-17:
When I tried @gustavmangion's suggestion, I realized that the official one is "Flashforge" and the custom one is "FlashForge". I tried changing it to "Flashforge" instead of "Flashforge (Custom)" or similar, and it works.

Turns out, this change is already on master: https://github.com/eugr/Flashforge-for-Cura/commit/d8dd67bb79419f221dc8c170f5cf1379afb6678e

So next release should include that fix. In the meantime, you can make the second "F" in "FlashForge" lowercase and it'll work.

aghzk324, 2024-02-24:
Hi it does not show up in cura version 5.6.0, macOS Sonoma 14.3. Any ideas? I have try rename the definations, also restart,...