eugr/Flashforge-for-Cura #31: Monoprice Dual (Creator X clone)  - Always prints with the Right Extruder

bthahn, 2023-04-13:
Cura 5.3

I am using this with my Monoprice Dual (Creator X clone) printer and it works great as long as I am printing from the right extruder.  The only issue is that it will ONLY print from the right extruder regardless of if I have 1 or 2 enabled and selected when I slice with Cura.  I disable the one I am not using prior to slicing but it makes no difference when I start the print it always prints with the right one..

cameronalred, 2023-05-15:
Wanted to add that I am also having this issue with my Creator Pro and latest Cura version.

eugr, 2023-05-17:
Upgraded to the latest Cura version and seeing the same behavior, will look into it

ukicomputers, 2023-06-18:
Same issue with Creator Pro

Trev4G, 2023-06-30:
I have this issue on Creator Pro but only on Cura 5.3.1 . After much installing discovered everything up until Cura 5.2.2 works ok. Also noticed that when changing extruders it is neccesary to delete the model and reload it. F5 does not work.

Trev4G, 2023-08-18:
Just go onto the Cura website and head for the software download. Underneath the "DOWNLOAD FOR FREE" is Find Previous versions" . Recommend 5.2.2. Just install no need to remove 5.4 they will exist together. You may need to get a copy of your settings from AppData\Roaming\cura and copy between versions.

nicklany01, 2024-12-25:
This PR by @alexbdv fixed this issue for me: 8117e78909471626d0420fbe8a8fe92510586d08

eugr, 2024-12-27:
I've merged the pull request, hopefully it's all resolved now.

closed 2024-12-27 by eugr