eugr/Flashforge-for-Cura #34: Cura 5.6 Creator Pro 2016_No preheating and wrong zero

jseifried, 2024-02-26:
Hi eugr,

at first thank you for your work!

Like described above my CreatorPro didnt start any preheating (bed and nozzle) and immediately start "printing" on the default printhead position right side next to the bed. It didnt drive to its preheat position (left side of bed), neither into zero position (middle of bed).

i dont know where is my failure. i copyed the files from definitions, extruders and meshes folders like described into local app data folders for cura5.6. Printer gets recognised and everything seems fine n cura. installed x3gwriter and restart cura. 

i just wonder a bit at the /src folder in your project. What to do with those .gcode files? May i need them for the x3gwriter? Their filenames sound like they are the missing link...

Best Regards J

Jamnoran, 2024-03-30:
I also get this issue.

Jamnoran, 2024-04-04:
I noticed it works on older versions, for example 4.4 worked for me.